RobertK66/obc_1769_core #12: Soft UART

RobertK66, 2022-03-05:
Switch the Debug Interface to use 2 general IO Pins and implement a
9600 baud Soft Uart

kodsurf, 2022-03-05:
I remmember we did that when working with TX RX on dev board :

[image]

And we added GPIO somewhere here :

[image]

This is the way how it is connected currently with 2 wires
[image]

Can you please tell which pins on OBC and RS485 to use ?

kodsurf, 2022-03-05:
Which pin to use ?

[image]

Have to change this in thruster initialization : 

[image]

Change to what ?

Thruster send byte function I change : 

[image]

In similar way ?

[image]

RobertK66, 2022-03-07:
Hi, I think your comment is releated to the [diskussion here.](https://github.com/RobertK66/obc_1769_core/discussions/5)
I also described the pin to use for RS485 there.

Thuis issue here is related in implementing a 'Soft UART' This means that code should make the timings to handkle 9600 baud RX and TX pins directly. There are some examples how to do this available but I had no time until now to implement it here......

RobertK66, 2023-04-09:
@krlorik : Hi, how is it going with this issue? Clear what to do?