RobertK66/obc_1769_core #15: GPS Enhancments

RobertK66, 2022-03-15:
- [ ] add support for PPS pin
- [ ] add power on/off support (enable pin)
- [ ] add basic PMTK command support (Tx UART conection)
     - [ ] get module type&version
     - [ ] choose NMEA packets to use and the period of transmission
     - [ ] make automatic Standby/On cycles available to use
- [ ] optional use of fix pin ?

RobertK66, 2022-09-27:
@kodsurf : Does the 'flight'-GPS have a PPS pin? 
This is a GPIO pin giving the exact timing of the communicated UTC timestamp. To have accuracy down to more then some 10/100 ms (which the NMEA msg take over the UART and UART buffers the received Chars and ...) you have to take the time from the msg and then 'wait' for the GPIO to make its transition (best by IRQ)....

RobertK66, 2023-02-07:
@WolfgangTreb , @AndiFHWN: any new info about 'Flight GPS' available (Type, techn. Spec, ....)?

AndyFHWN, 2023-02-07:
The decision on GPS is coupled with ADCS, because we do not need an extra module if GPS is integrated in future ADCS. Sorry, most likely this will be delayed.

RobertK66, 2023-02-07:
No Prob, I only try to make the issue list some sort of 'complete', because for the first time in this project my head has a approximate idea what a 'final OBC-Core' Firmware could look like :-D.

RobertK66, 2023-02-07:
And I use the 'question' tag now to indicate the ones we (OBC developer) have to wait for info....