RobertK66/obc_1769_core #24: Configure, check and test all3 I2C buses on EM2

RobertK66, 2022-06-03:
- Enable all 4 sidepanel I2C buses and configre them (to use 100kHz at this time)
- write testcode to address external slaves

kodsurf, 2022-06-08:
During initialization of LPC_I2C

[image]

I enabled switches 

which should activate LPC_I2C2 on A and B sides
[image]

[image]

Then I checked pinning_climb.h
[image]

All those switches are by default initialized with **HIGH** value.    

Meaning that all switches are enabled by default ? 

However the strange thing that I noticed : 

[image]

If I manually set those switches to LOW  -  I2C still works (with wrong bytes)

RobertK66, 2022-06-19:
Remark: You can test this (switch every Enable Pin) with my hw_check command:

| 'h'     | \<pinIdx\> \<mode\>       | sets an GPIO pin to mode (0: initVal, 1: high, 2: low, 3: slow blink, 4: fast osz.)                            |

The Pin Idx for the 4 I2C EN pins are: 

	PINIDX_I2C_A_EN  31
	PINIDX_I2C_D_EN  32
	PINIDX_I2C_C_EN  33
	PINIDX_I2C_B_EN  42

kodsurf, 2022-06-22:
After latest remark regarding enabling C/D in  pinning_climbobc.h : 

I2C works correctly on every sidepanel.

Switches also work as expected.

closed 2022-06-22 by kodsurf