RobertK66/obc_1769_core #25: Implement thruster "Test Fire" from OBC

kodsurf, 2022-06-28:
Implement code for performing "test fire" for thruster simulator hardware. 

Sequence : 
1) Version request for a thruster module during initialization, to verify that thruster subsystem is connected and ready

2) Set thrust and specific impulse to an value (example: 350microN, 3500s)

3) Set thruster mode to start firing sequence ( which initialize heating of reservoire)

4) Firing will occure when reservoire would be heated to sufficient temperature. Monitor the temperature while its heated up. Parse temperature and thruster status values out of register map.

5) With sufficient temperature thruster should change status to firing. 

6) After fire for defined time - idle.

Plan is to implement it in layer_7_app. With start/stop/set/read value requests sent manually from command interface.

Note :  1)**thruster replies "instantly" after request is sent** 
**2) Length of the reply message is constant and depend on type of the request.** 
3) There are no unique bytes which would indicate the beggining of reply message, however reply length is know. Reply is only sent upon request. Reply is sent "instantly" after request.

Keeping "Notes" in mind - design RX buffer handling.  For every specific request there should be corresponding function that proccesses the received buffer.

Example :  Read version request is transmitted ---> expected RX buffer length is N bytes.  ----> check validity of reply message with checksum ---> parse version number ----> compare with expected version to verify that correct hardware is connected.

Example 2:  Set Thruster Mode request is transmitted ---> expected RX buffer length is N bytes.  ----> check validity of reply message with checksum ---> Thruster should reply that mode is succesfully changed. Parse this information ----> return ok status

RobertK66, 2022-06-29:
I think it’s a good idea to implement all this test commands in 'layer_7_app' for a beginning. What comes to my mind here is, that it maybe would be a good idea to separate all 'sequence timings' (if there are some) into a new 'Lx_thruster' module (where x is something between 3 and 7 😄 ) with this separation of concern:
- layer_7_app:
all test commands with specific values and defaults, all responses to these commands translated to debug_interface 'events'
- l4_thrctrl:
timings and status changes which have to be done automatically (e.g. wait for heating and go to 'fire' mode, ...)
- thr (existing file):
low level Rx TX buffers and immediate reactions to RS485 answers from thruster.           

@kodsurf: pls. indicate if you start with 'real work' on this issue by assigning yourself to this issue in github (available somewhere on the right side of the github web-interface)
to all: My 'git culture' is: Nobody ever assign somebody else to a task. Everybody has to 'self assign'. This gives some basic commitment, that somebody already is 'in charge' on advancing the resolution of the issue. If you have reasons to stop your involvement then pls. ‘de-assign’ (and make a short comment about status and reasons)

RobertK66, 2022-06-29:
we have this already see https://github.com/RobertK66/obc_1769_core/blob/develop/doc/m03_systime.md

RobertK66, 2022-06-29:
for commands you can still use app_l7, no prob. I meant to introduce another #level# for your thruster specific timing and states (e.g. wait for temp and so on...) then you would have your code on 3 different layers:
APP:    code for (test/debug) commanding only
L4 (new):  code for coarse timing and mode control
L2/3 (existing thr.c?):    Rs485 timings and timeouts if there is no answer from RS485....

kodsurf, 2022-09-09:
I made a skeleton function for Thruster Firing Sequence that seems working ok

[image]

Firing sequence is triggered by debug command "8 5000"  where 5000 is ms duration of fire.

Begin and Stop timestamps correspond to what was expected.

[image]

kodsurf, 2022-09-09:
[image]

Sometimes it gets into infinite loop situation where whole OBC stucks and needs reset. 

Something wrong

kodsurf, 2022-09-09:
Please review feature/thruster-fire

at latest commit https://github.com/RobertK66/obc_1769_core/commit/6ce230d061eb0e4e7b9e85a16471249770ca4bce

kodsurf, 2022-09-11:
I don't like previous approach that I attempted.

I think better way to do it is to create array of function pointers.

Let's call it "thruster_execution_sequence_stack"

Then we create set of basic function with arguments argc,argc. Asside to already existing functions for thruster request. 

We assemble execution stack based on small functions.

kodsurf, 2022-09-11:
Since communication with ground is only around 3 times a week.

It is imposible to control space craft continuously .

During communication window - instructions to assemble the execution stack + conditions/ timings could be transmitted.

In this way it would be possible for example to transmit execution sequence to Space Craft which would be either executed on specific time step or when right conditions occured ( gps, attitude, battery power, etc)

"Conditions" are global variables that are monitored and updated

kodsurf, 2022-09-11:
At the latest commit of current branch https://github.com/RobertK66/obc_1769_core/commit/d9ed87af60bc60f480b4b44c2e7eb30ad1b00e6e

I completed a prototype skeleton for stack sequence execution

[image]

with thr_wait() functions applied with different parameters - it works well. 

However if I use other thruster functions that prints bytes to RS485 - there are problems

kodsurf, 2022-09-11:
I definetly need an advice here ! 

[image]

if waiting time between stages (stack actions) is to small. OBC can go into hard_failure() mode !!!!

This has to do with timings of multiple modules.

But in general my approach worked - stack sequence (including thruster requests) was executed as expected

kodsurf, 2022-09-13:
@RobertK66  I definetly need your code review at this stage. (do verify that what I did is not completely stupid)

There is no need in thruster hw connected to test out sequence execution module. Therefore code should be runnable from your side

To trigger sequence execution use command 8 <sequence index>  (which is 0,1,2  as only 3 test sequences programmed)

Sequeces are programmed in init of module

[image]
0-th argv of functions in sequence should be hardcoded to <sequence index>

I achieved that multiple seqeunces can run in paralel. 

But problem is that sometimes 0-th sequence leads to hard failure infinite loop. I am strugling to understand why ! Pls review

RobertK66, 2022-09-13:
ok, give me 2 days pls....

kodsurf, 2022-09-13:
I violated my own rule of "not developing several different features into the same branch"

But either way I would need memory read and write to save thruster register data into the log files (later on).

Therefore I would leave those memory read write skeleton functions here.  Tested, understood how they work . 

[image]

https://github.com/RobertK66/obc_1769_core/commit/718d533c68cd97c78cf1cf5897284037d4052dfa

RobertK66, 2022-11-24:
fixed with #59

closed 2022-11-24 by RobertK66