RobertK66/obc_1769_core #31: Fix git issues with "STP" branch

kodsurf, 2022-08-03:
Due to previously not well understood project maintenance with git  (by me) : STP branch is million commits behind dev/master. 

Need to synchronize "develop" with "STP"  by implementing all the relevant STP features into dev and make pull request.

Further on to avoid this mess - I will be doing feature/< NAME >   branch for every new module.  And Pull Request as soon as module is finalized.

RobertK66, 2022-08-09:
You can try to merge develop  >  STP. git normally gives some good starting point to get the differences resolved when same files are changed on both sides.....

RobertK66, 2022-08-09:
I can show you how this works (for #36 after we successfully pull your #34)

RobertK66, 2022-08-16:
ok unfortunatly this was a desaster 🤪 . I managed after our session. Will prepare some doku on how to do this with eclipse.
generated new #37 for this purpose now.

RobertK66, 2022-09-27:
@kodsurf : can we close this? is there something useful left in the STP branch?

kodsurf, 2022-09-27:
Nonono Please don't close STP branch for now ! I2c is not yet integrated! And all my usable code is there

I wanted to integrate i2c to dev branch with actual psu code. And not test Arduino. That is why I did not proceed with i2c dev integration yet.

RobertK66, 2023-02-07:
@kodsurf : something more to do here?
(The branch will not be touched by closing this issue. So pls. close it if nothing else is to be done here.

RobertK66, 2024-06-13:
stalled and @kodsurf no longer available in proj?

closed 2024-06-13 by RobertK66

kodsurf, 2024-06-13:
Depends on if someone wants to hire me for that :)