RobertK66/obc_1769_core #33: RadTest II: Readout sensor values

RobertK66, 2022-08-04:


kodsurf, 2022-08-10:
[image]
By using this in layer 3sensor ?

readJob.rx_data should be printed to debug uart with system events string every 10 seconds ?

kodsurf, 2022-08-10:
https://github.com/RobertK66/obc_1769_core/pull/36/commits/eea7990b273b370638fc188713a1e64ca7635e75

Is this what was meant for read sensor values ?

why not combine and continue with it in issue32 ?

RobertK66, 2022-08-11:
Yes this should be ok. 
you mean to add it to https://github.com/RobertK66/obc_1769_core/tree/feature/issue32 branch and together with #36 pull request? Yes seems feasible because ;-).

RobertK66, 2022-08-16:
fixed with #32

closed 2022-08-16 by RobertK66