RobertK66/obc_1769_core #40: Only utf-8 should be printed to debug uart for radtest

kodsurf, 2022-08-16:
All characters that are sent over debug uart has to be utf-8 (ascii)

Non ascii charachet received by rasbery pi would trigger "try, except" failed to read com port error
[image]

Log file from raspberry pi shows this error catched by python try except
[image]

kodsurf, 2022-08-16:
[image]

I checked the latest "developer" branch code merged by robert.

All output is indeed utf-8 now. 

We will reflash the OBC again tommorow (17.08) and try to see if error fixed. 

P.S - Looks like pure_print_debug() without non ascii  event id bytes was actually neccessary to be written :)

kodsurf, 2022-08-16:
@RobertK66  can you please confirm with your updates regarding sensor output code that there will be no utf-8 symbols ?

Otherwise we would have to change python code to proccess binary data. Which is much more complicated then changing it on OBC side...

RobertK66, 2022-08-17:
Yes I did this when merging the code with your pure_print_debug() I found. The Magic is done in line 127-147 of L7_climb_app.c

We now have the rtst_eventoutput(event_t event) function where we can translate events to pure ASCII strings output for the radtest

RobertK66, 2022-08-17:
I also remember how we did the analyses after the tests last time:

Each line with some usefull values (e.g. the Sensor Line) is now formated as ';' seperated string. This together with the header (output with rtst_init() ) can be used to be simply loaded into an Excel where the values then can be analized -> into diagrams or whatever. 

So I plan to use this format also for the 'results' (Counters,bit error counters ) of the memory tests. The first field is the 'type pf record' text.

kodsurf, 2022-08-17:
Here are the log photo from test carried out by Abbas today, with latest develop branch code

[image]

@RobertK66  Please verify that debug uart works as intended for the test.

RobertK66, 2022-08-17:
I can't verify that ;-). I never saw the screen you show here. I only got log files from Andi after the test, and I analysed my memory stuff with Excel then.

Do you have text files where you can read all the sensor values written afterwards? Are the values feasable (temp, voltage, current)?

RobertK66, 2022-08-17:
I think this is okay already with current developer branch.

closed 2022-08-17 by RobertK66