RobertK66/obc_1769_core #44: dev branch project is not compilable git clone

kodsurf, 2022-09-06:
After git clone and submodules init

[image]

Project remains uncompiliable due to folder structure of ado_chip_175x_6x
[image]

To make project compilable I have to manually copy files from ado_chip_175x_6x folder. But this way git references are removed for ado project.

kodsurf, 2022-09-06:
[image]

If manually copy ado files - it creates mess with untracked files for git

kodsurf, 2022-09-06:
https://github.com/RobertK66/ado-chip-175x-6x/commit/66bce532e3781dd6291068f80021379024f23d97

This commit with CRC8 checksum functions is not on ado dev branch after clonning  repo

Strange .....

kodsurf, 2022-09-06:
Changes from the commit 

https://github.com/RobertK66/ado-chip-175x-6x/commit/66bce532e3781dd6291068f80021379024f23d97

are not listed anywhere on ado dev branch !!! 

I manually added changes based on diff of commit. But changes are not on git repo ! Veeery strange ..

RobertK66, 2022-09-06:
Hi, sorry I started to write up some Doku about this submodule/checkout issue(s). Unfortunately I am not ready yet. I try to look into this tomorrow afternoon. Quick hint: Submodule (and 'Update Submodule) always refers to a single commit (should be good enough for you), BUT you can have the eclipse to 'sit' on a branch and checkout, if you (mainly its me ;-)) want to be able to develop and commit in lib module developer branch. Again I did not fully understand Eclipse client here but hope to get it sorted out later this week .. (I am available for Skype/Teams earliest Thursday, sorry....)

RobertK66, 2022-09-18:
Stil working on the doku point for that....

RobertK66, 2022-12-03:
I am now thinking of re-join the ado code into the obc_core code repository again....
(When I introduced the ado lib repository being an external repository and using this submodule thing, I had the idea that I will use it for different hardware projects with this CPU. As this is no longer the case with me, rejoining the code without submodule will be simple way for all to avoid this confusions.....

RobertK66, 2022-12-03:
I will do this next time we change or fix something in the ado lib code.

commit 3b29755b (2023-01-08): Merge pull request #66 from RobertK66/feature/i17_anothertry

fix #44
closed 2023-01-08 by RobertK66