RobertK66/obc_1769_core #47: Create python script for orbital position prediction based on TLE

kodsurf, 2022-09-27:
1)  TLE can be obtained by API

https://celestrak.org/NORAD/elements/gp.php?CATNR=25544&FORMAT=JSON

CATNR is ID of object in NORAD database

Object ID can be searched by the name using 
https://celestrak.org/satcat/search.php

Example : 

[image]

[image]

substitute CARNR with PEGASUS ID : 

https://celestrak.org/NORAD/elements/gp.php?CATNR=42784&FORMAT=JSON

To obtain TLE of PEGASUS 

JSON
[image]

Or text

https://celestrak.org/NORAD/elements/gp.php?CATNR=42784&FORMAT=TLE

Output of simple HTTP GET request is 

[image]

3 lines of strings. 

2) Capture those strings with python  (using curl http request)

And use : https://rhodesmill.org/pyephem/    

pyephem to calculate object possition (lat/lon/alt)  

Example : 
https://space.stackexchange.com/questions/4211/calculate-satellite-coordinates-from-tle-data

[image]

3) Use obtained lat,lon,alt ( and possible other data)

to create NMEA messages and print them to UART in a loop over time.

kodsurf, 2022-10-31:
[image]

I added python script to simulate NMEA over orbit: https://github.com/RobertK66/obc_1769_core/commit/9c10894b416563b53adac0c0dfe166f4c7dabbe6

currently only for RMC message (will add another messages latter)

Looks ok but some unnessesary bytes received 
[image]

kodsurf, 2022-10-31:
I struggle to understand how OBC NMEA parser and checksum calculation works

Example : I take a valid nmea message from 
https://orolia.com/manuals/VSP/Content/NC_and_SS/Com/Topics/APPENDIX/NMEA_GGAmess.htm
`"$GPGGA,123519.00,4807.038,N,01131.000,E,1,08,0.9,545.4,M,-164.0,M,,,,*47"`

Then I simply print out this message to COM port in the loop every 1s
[image]

What I receive on OBC is SysEvent String 
[image]

Why so ?

kodsurf, 2022-11-01:
[image]

looks ok to me

kodsurf, 2022-11-03:
Only one thing left before I mark this python task as completed. 

[image]

With UTC timestamp that I supply for RMC message - I never reach  line 348 because  day < 1

I compile RMC message according to https://orolia.com/manuals/VSP/Content/NC_and_SS/Com/Topics/APPENDIX/NMEA_RMCmess.htm

[image]

example of MY  string of NMEA RMC message

[image]

Maybe this is not what OBC RMC parser expects ?

This is how I compile UTC string

input : 
yy - year (2022)
mon - month (11)
dd - date (3)
hh - hours (6)
mm -min (30)
ss - sec (5)
[image]

I checked OBC code 

[image]

Looks like It is the same format that I supply.  But why is then day is wrong ?

kodsurf, 2022-11-03:
Error is in date compilation of RMC message in python script.

"1102022" - WRONG

Correct format is ddmmyy - 03112022 ( or 3112022???)

What if it is first January?
01012023 or 112023

kodsurf, 2022-11-03:
http://www.ae.utexas.edu/courses/ase389p7/projects/svatek/commands/RMC.html

Jevgeni please use Google before posting something stupid on issues.

Correct nmea message format with description is given at link above.

3.11.2022 is 031122

kodsurf, 2022-11-03:
[image]

https://github.com/RobertK66/obc_1769_core/commit/350f1099ea445e6fe7cfdec55f9cdce1ad13262f

I fixed date compilation in RMC message. Now everything works correctly as it seems with RMC parser of OBC.

RobertK66, 2022-11-26:
Is the Script available somewhere?
You could share it in the tool folder, if you want.....

kodsurf, 2022-11-26:
It is on feature/gps-integrate 

Utilities/python scripts

RobertK66, 2022-11-29:
I will keep only the long lat values in OBC code and integrate this into develop without the debug stuff....

commit 4a8642e6 (2022-11-29): Merge pull request #61 from RobertK66/feature/gps-integrate

Feature/gps integrate
fix #47
closed 2022-11-29 by RobertK66