RobertK66/obc_1769_core #53: Checksum calculation for NMEA message parser does not work properly

kodsurf, 2022-11-01:
Either I do something wrong, or 

[image]

Those checksum cases does not do what they are expected to do. 

I had to manually turn off checsum check in order to proccessing function to be executed : https://github.com/RobertK66/obc_1769_core/commit/5c1a7d711af5fe2c5be69769a102eca3c48a1f8a

P.S : This is how I calculate checksum for autogenerated NMEA string : https://electronics.stackexchange.com/questions/214278/generating-hexadecimal-checksum-for-a-nmea-pmtk-message

If I take example NMEA message : $GPGGA,115739.00,4158.8441367,N,09147.4416929,W,4,13,0.9,255.747,M,-32.00,M,01,0000*6E

(with valid checksum from google  )  - Original dev branch NMEA parser still gives an error.

kodsurf, 2022-11-02:
https://github.com/RobertK66/obc_1769_core/commit/7c087acd96c415e94a92e56667582c03c48987e7

Check summ is no ok... There was no problem with checksum.  Its just stupid me. But I solve it

kodsurf, 2022-11-02:
But there is something even more stupid that I have discovered accidently. 

I stupidly left some int r =4 ;     for some reasons I dont remember and forgot to delete it. 
[image]

The thing is ....  when I tried to cleanup the code after checksum problem cleared up - I deviced to delete all rubish from code .... 

BUT WHEN I DELETE STUPID int r=4;

I get ... Hard fault . And not instantly, after arround 10-20 sec of parser working ...

[image]

LAST_STARTED_MODULE = 3 

.... what a ....

kodsurf, 2022-11-03:
Sorry no problem there .... 

It was just me stupid. 

Resolved

closed 2022-11-03 by kodsurf

RobertK66, 2022-11-07:
No reason to apologize. I was just asking. Thats what Issues are for (initiate communication to learn something about the issue).