RobertK66/obc_1769_core #57: Thruster module: Switch text outputs to Binary Event API

RobertK66, 2022-11-14:
At least remove the usage of the  deb_print_pure_debug() function and replace with 'STR' or 'RAW' event. 

But in the end the Thruster module should provide a binary event API. See e.g. this output:

```
[07:32:12] app/str :' Sequence id=0 start '                                                                                           
[07:32:12] app/str :' Stage SET index= 0 completed '                                                                          
[07:32:12] app/str :' Action 31001:Set Operational Mode 0 '                                                              
[07:32:12] app/str :' Stage SET index= 1 completed '                                                                         
[07:32:12] app/str :' Action 31002: Set Emitter Mode 0 '                                                                    
[07:32:12] app/str :' Stage SET index= 2 completed '                                                                         
[07:32:12] app/str :' Action 31003:Set Extractor Mode 0 '                                                                  
[07:32:12] app/str :' Stage SET index= 3 completed '                                                                         
[07:32:12] app/str :' Action 31004: Set Neutralizer Mode 0 '                                                              
[07:32:12] app/str :' Stage SET index= 4 completed '                                                                         
[07:32:12] app/str :' Action 31005: Set Temperature Mode 0 '                                                           
[07:32:12] app/str :' Stage SET index= 5 completed '                                                                         
[07:32:12] app/str :' Action 31006: Set Reservoire Heater Mode 0 '                                                   
[07:32:12] app/str :' Stage SET index= 6 completed '                                                                         
[07:32:12] app/str :' Action 31007: Set Thrust Ref 0 N '                                                                      
[07:32:12] app/str :' Stage SET index= 7 completed '                                                                         
[07:32:12] app/str :' Action 31008: Set Reservoir Heater Power Ref 6 W '                                          
[07:32:12] app/str :' Stage SET index= 8 completed '                                                                         
[07:32:12] app/str :' Action 31008: Reservoir Heater Power Ref 6 W '                                                
[07:32:12] app/str :' Sequence complete '                                                                                                  

The strings running out over our debug UART here are aprx. 600 bytes sent by 9600 baud -> taking up aprx 65 millSeconds.

The real data included can be expressed as pure 6bit Event IDs for (Seq-started, seq-stoped) and some events with 1/2/4 byte data (Set-Stage-IDX (idx), Action-Nr (Nr), Set-Mode(type, mode), Set-Power-Ref(power). Sent as binary events only I thiong this will be 
2*2 + 9*3 + 9*4 = 70 bytes -> only blocking the debug TX UART for 7,3mS !!! So its a performance factor of aprox. x10 on the debug uart.....

kodsurf, 2022-11-14:
Can you then please share the tool that you use to readout binary debug messages ?

Please one line code example on how to print as app/str event ( I think you already did it)

RobertK66, 2022-11-14:
You can find the latest 'release' here: https://github.com/RobertK66/embedded-toolbox/releases 
A release can be used with the provided download without IDE from command line. (Should work both in windows and in linux ...)

This is - as mentioned - my 'experimental Stuff' playing around with dotnet. You can try to build from sources with any DOTNET IDE. I do use Visual Studio (classic) and not 'Visual Studio Code". But it should be possible to build with either .....

If you use the binary downloaded version you can adapt the UART settings in appsettings.json file. There you also find the definition of OBC events (from the time of that release. If you use already coded/used types (uint8/uint16 usw.) you should be able to adapt this config to refect new events / modules coded in OBC.....