RobertK66/obc_1769_core #67: Not able to compile develop branch

kodsurf, 2023-01-15:
Please fix compilation error with dev branch. MCUExpresso runs crazy when I pull recent commits

kodsurf, 2023-01-15:
[image]

RobertK66, 2023-01-16:
ok, what I did is, ichanged the project layout. Both projects needed by MCUXpresso ("ClimbObc" and "ado_chip_175x_6x") are now  on the same level under the (git) root folder. I think you have to close the projects and reopen it from the correct position (or make a new workspace and clean checkout/import......

I also overlooked this when my github actions tried and failed to build. Will fix this soon.....

RobertK66, 2023-01-16:
as you can see here: https://github.com/RobertK66/obc_1769_core/actions/runs/3928554714/jobs/6716301653 the develop branch is able to build without warnings and errors. ()I only had to adapt the project path in the action scripts. No change in Source codes and project files was needed. 
So pls. make a clear new workspace and you should be ok then.....

RobertK66, 2023-02-07:
@kodsurf : something new on that one?

In order to not get others confused about this, I close it now.

closed 2023-02-07 by RobertK66