RobertK66/obc_1769_core #72: Which sidepanel GPS ?

kodsurf, 2023-02-08:
In a current develop branch gps module is initialized on LPC_UART0. However comment on the code says that it is sidepanel C  (X-)

[image]
This looks incorrect.

My own docu "originaly" stated that LPC_UART0 is C/X-
[image]

Yesterday with flatbed setup it turned out that LPC_UART3 is C/X-

-------------------------------------------------------------------------

Please clarify to which sidepanel GPS should be connected by design, and check the code for gps init data.

kodsurf, 2023-02-08:
Ok, based on your adcs conversation, gps data to be supplied from ADCS. Meaning that climb_gps.c module will likely be deprecated.

If is ok - I will for now change gps UART to LPC_UART3 with direct commit to dev ?

RobertK66, 2023-02-09:
My GPS code was based on (my!) assumption that we will need/have any GPS for time sync. The init structure allows for 2 additional IO pins which the GPS type I used provides. Code does not make use of it until now (see #15).

About 'the real thing' pls. refer to available CLIMB hardware design and Doku. The last known SatBusDefinition_v01_Table.pdf I have (4.8.2020!) does show UART usage: A-C: Stacie-A, Stacie-C B: S-Stacie D: thruster. What I remember is, that it was decided definitively to use a soft Uart (2 GPIOs) for Debug interface in future HW revisions - #12  (but I don't know where this is documented).

@WolfgangTreb , @AndyFHWN : Pls. clarify status of current design and hardware documentation!