RobertK66/obc_1769_core #75: CppEthernet

krlorik, 2023-02-20:
Downloading the lpc1769 in the Procedure it is written to create a new folder with CppEthernet, but it seems so that a folder with that name exist already in the project. Screenshot will follow as better description of the issue.

RobertK66, 2023-02-21:
CppEthernet !?
I think you refer to another repos here!? ( https://github.com/RobertK66/Climb-Obc-Sandbox ). I referred this as reference to 'what  happend earlier', but I did not visit it for some time now.

Which 'Procedure' do you refer here? Some How To from that repos?

RobertK66, 2023-02-21:
For using this actual repos here you should prepare a new empty workspace where after git checkout you should get the 2 projects named 'ado_chip_175x_6x' and 'ClimbObc'.